AdvancedClimateSystems/uModbus #51: Modbus RTU over TCP

ccatterina, 2017-09-19:
Hi,
Is there a way to send RTU messages through TCP?
I need this functionality because we usually convert all the serial device with rs485/ethernet converter.

OrangeTux, 2017-09-19:
From what I understand you want have a Modbus RTU Master that needs to send requests to Modbus TCP slave. Is that correct?

ccatterina, 2017-09-19:
No, i have a Modbus TCP master (my raspberrypi) that needs to send requests to a Modbus RTU slave, in the middle there is a rs485/ethernet converter that receives the requests from the raspberry and sends them to the slave through the serial line.

OrangeTux, 2017-09-19:
This is no problem. I've build one myself using the building blocks that uModbus provides. 

As you may know Modbus RTU and Modbus TCP have a few bytes in common, the PDU. As shown in [this image](http://www.simplymodbus.ca/images/adu_pdu.PNG).

To convert a Modbus TCP request into a Modbus RTU request you need to strip the MBAP header and add a CRC. After you receive the RTU response you need remove the CRC and prepend the message with the MBAP header and return that.

ccatterina, 2017-09-20:
Thank you very much. That's all I needed.

I close the iusse.

closed 2017-09-20 by ccatterina

rahimdzb, 2020-10-29:
Does

> D'après ce que j'ai compris, vous avez un maître Modbus RTU qui doit envoyer des requêtes à l'esclave Modbus TCP. Est-ce exacte ?
In this case what we do exactly !