AdvancedClimateSystems/uModbus #60: pip3 install

joaopmrod, 2018-06-09:
I get this error when trying to install on my odroid-xu4 (Ubuntu 18.04 LTS (GNU/Linux 4.14.37-135 armv7l, Python 3.6.5 (default, Apr  1 2018, 05:46:30) [GCC 7.3.0] on linux)

How can I fix it?
```
odroid@odroid:~/centopeia_firm$ pip3 install umodbus
Collecting umodbus
  Using cached https://files.pythonhosted.org/packages/9d/ef/a575ba0840854b937ab7e5c6c4b40c4491089e42114a9b4413e87a84b2a8/uModbus-1.0.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-ty56b9mk/umodbus/setup.py", line 12, in <module>
        long_description = open(os.path.join(cwd, 'README.rst'), 'r').read()
      File "/usr/lib/python3.6/encodings/ascii.py", line 26, in decode
        return codecs.ascii_decode(input, self.errors)[0]
    UnicodeDecodeError: 'ascii' codec can't decode byte 0xce in position 547: ordinal not in range(128)

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-ty56b9mk/umodbus/
```

OrangeTux, 2018-06-10:
Thanks for submitting this issue. At the moment I don't have much time to look into it. But a quick workaround for you would be downloading the source and then run `python setup.py` to install it manually.

OrangeTux, 2018-09-10:
It seems that you are having network issues unrelated to uModbus.

8minutenergy, 2018-09-11:
I am running into this exact same issue building a Linux container image for docker.
pip install works fine
pip3 install fails with 

```
Collecting umodbus
  Downloading https://files.pythonhosted.org/packages/9d/ef/a575ba0840854b937ab7e5c6c4b40c4491089e42114a9b4413e87a84b2a8/uModbus-1.0.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-3p0i_lr3/umodbus/setup.py", line 12, in <module>
        long_description = open(os.path.join(cwd, 'README.rst'), 'r').read()
      File "/usr/lib/python3.6/encodings/ascii.py", line 26, in decode
        return codecs.ascii_decode(input, self.errors)[0]
    UnicodeDecodeError: 'ascii' codec can't decode byte 0xce in position 547: ordinal not in range(128)

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-3p0i_lr3/umodbus/
```

OrangeTux, 2018-09-12:
@8minutenergy I cannot reproduce the bug. Could you post which docker images you used and which commands you issued?

8minutenergy, 2018-09-13:
I am using ubuntu:cosmic. Work around is to run pip install umodbus instead of pip3 install umodbus.

Dockerfile 
----------

FROM ubuntu:cosmic

SHELL [ "bash", "-c" ]

RUN apt-get update && apt-get install -y tzdata && apt-get install -y \
    build-essential \
    python-dev \
    python-setuptools \
    openssl \
    libssl-dev \
    libevent-dev \
    python-pip \
    git \
    gnupg \
    dirmngr \
    libyaml-dev \
    openssh-client \
    python3 \
    python3-dev \
    python3-venv \
    python3-pip \
    python3-setuptools \
    python-numpy \
    python-pandas \
    && pip install PyYAML \
    && pip install umodbus \
    && pip3 install click \
    && pip3 install voluptuous \
    && pip3 install umodbus \
    && rm -rf /var/lib/apt/lists/*

OrangeTux, 2018-09-14:
Weird, It works on my machine.... I'll later investigate it later

OrangeTux, 2018-09-14:
@8minutenergy Can you add this line to the Dockerfile:

```
FROM ubuntu:cosmic

ENV LANG C.UTF-8

SHELL [ "bash", "-c" ]

RUN apt-get update && apt-get install -y tzdata && apt-get install -y \
        build-essential \
        python-dev \
        python-setuptools \
        openssl \
        libssl-dev \
        libevent-dev \
        python-pip \
        git \
        gnupg \
        dirmngr \
        libyaml-dev \
        openssh-client \
        python3 \
        python3-dev \
        python3-venv \
        python3-pip \
        python3-setuptools \
        python-numpy \
        python-pandas \
        && pip install PyYAML \
        && pip install umodbus \
        && pip3 install click \
        && pip3 install voluptuous \
        && pip3 install umodbus \
        && rm -rf /var/lib/apt/lists/*
```

OrangeTux, 2018-11-05:
Closed due to inactivity. You can reopen if you feel the need

closed 2018-11-05 by OrangeTux