AdvancedClimateSystems/uModbus #61: ReadHoldingRegisters dosen't support Long/Float data type?

lxcsmallcity, 2018-06-30:
ReadHoldingRegisters dosen't support Long/Float data type?

And,Float data type contained: "AB CD","CD AB" data format

OrangeTux, 2018-06-30:
> ReadHoldingRegisters dosen't support Long/Float data type?

No, uModbus only supports 16 bits integers. Those are unsigned by default, but can be signed. Read [here](http://umodbus.readthedocs.io/en/latest/configuration.html) for more on that.

If you want read or write  floats according to [IEEE 754](https://en.wikipedia.org/wiki/Single-precision_floating-point_format) you need to split the float into 2 16 bit numbers. 

> And,Float data type contained: "AB CD","CD AB" data format

I'm not sure what you mean by this?

lxcsmallcity, 2018-07-01:
Yes,I want read or write floats according to IEEE 754.Like "electric instrument" devices etc.

OrangeTux, 2018-07-02:
Well, to write binary32 number to 2 registers you can do this::

``` python
import struct

f = -123.456

# IEE 754 representation of f.
binary32 = struct.pack('>d', f)

high_byte, low_byte = struct.unpack('>HH', binary32)
```
Now you can write the high and low byte on 2 separate registers.

To read 2 registers and combine them back to a float you can do this:

``` python

import struct
high_byte = 49910
low_byte = 59769

# f is -123.45600128173828
f = struct.unpack('>f', struct.pack('>H', high_byte) + struct.pack('>H', low_byte)[0]
```

lxcsmallcity, 2018-07-04:
OK,Thank you very much.
Have you think about add the function into next version?
Like "modbus_tk" library,it can be input parameter “ data_format='>f' ”

OrangeTux, 2018-07-04:
No I'm not considering this right now.  For now I think the increase of API complexity doesn't weight out the gains in functionality.  

And as I demonstrated, it can easily be done with in the existing API.

closed 2018-07-04 by OrangeTux