AdvancedClimateSystems/uModbus #62: start client

anotherkeycode, 2018-10-19:
I can't start the client example.

Python 3.7.0 (default, Sep 15 2018, 19:13:07) 
[GCC 8.2.1 20180831] on linux

[dev@dev-vm modbus]$ python modbus_client.py 
Traceback (most recent call last):
  File "modbus_client.py", line 18, in <module>
    response = tcp.send_message(message, sock)
  File "/usr/lib/python3.7/site-packages/umodbus/client/tcp.py", line 262, in send_message
    raise_for_exception_adu(response_error_adu)
  File "/usr/lib/python3.7/site-packages/umodbus/client/tcp.py", line 247, in raise_for_exception_adu
    pdu_to_function_code_or_raise_error(resp_pdu)
  File "/usr/lib/python3.7/site-packages/umodbus/functions.py", line 111, in pdu_to_function_code_or_raise_error
    raise error_code_to_exception_map[error_code]
umodbus.exceptions.IllegalDataAddressError:  The data address received in de request is not an allowable address for
    the server.

How can I fix it?

OrangeTux, 2018-10-24:
Thanks for you question!

The server responds with an `IllegalDataAddressError`. So you're requesting a coil or register which does't exists in the server. 

Oh, and could use wrap the code snippet with '```'. This will apply some markup to the code to make it more readable. Read more about it [here](https://help.github.com/articles/creating-and-highlighting-code-blocks/).

closed 2018-11-05 by OrangeTux

codevak, 2020-06-12:
As a newbie to modbus protocol, I am facing the same issue of `umodbus.exceptions.IllegalDataAddressError: The data address received in de request is not an allowable address for the server`. 
Can the client and server example in the repo for TCP be used as is? If not, how should they be modified to work together? I am trying to simulate both the client and server using the python code given in the repo.

OrangeTux, 2020-06-15:
> Can the client and server example in the repo for TCP be used as is?

Yes, they can work together.

codevak, 2020-06-15:
Hi, Thanks for your reply. 
I tried it already but end up with this error:` umodbus.exceptions.IllegalDataAddressError: The data address received in the request is not an allowable address for the server`
I start the server first (which is listening for incoming connections). Then I start the client in a different terminal. The client throws this error and stops. Any idea what I could be doing wrong?

albcp, 2020-06-25:
Hi @codevak, the code for the server included in the Readme is wrong, you need to modify _function_codes=[3, 4]_ to _function_codes=[1, 2]_ and _function_codes=[6, 16]_ to _function_codes=[5, 15]_.

codevak, 2020-06-26:
Hi, This works now. Thanks a lot!