AdvancedClimateSystems/uModbus #69: have util.pack_exception_pdu error when function_code is bigger then 16

Blacklagoon21, 2019-04-04:
In function pack_exception_pdu, when function_code is bigger then 16, struct.pack will error return.

    :param error_code: Error code.
    :param function_code: Function code.
    :return: PDU of 2 bytes.
    """
    return struct.pack('>BB', function_code + 0x80, error_code)

jaapz, 2019-04-04:
According to the modbus specification, the function code should never be larger than 16. How do you propose we fix this? Maybe we should just validate whether the function code is in the list of modbus defined function codes and raise an exception when it's not?

Blacklagoon21, 2019-04-05:
I tried to make same situation for the error. and I realize that choose wrong modbus mode (it was RTU over TCP/IP). and it works on modbus TCP/IP and without any error.

closed 2019-04-05 by Blacklagoon21