AdvancedClimateSystems/uModbus #78: Add support for encryption

cigarplug, 2019-11-30:
Hi!

I was wondering if it's desirable to add support for encrypting messages between the client and server. It could be made optional, and off by default. I took a course related to network security recently and did something similar as a part of an assignment, which means some groundwork has already been laid. 
I'd be happy to create a pull request. Do let me know your thoughts on this.

Cheers

OrangeTux, 2019-12-03:
Hello @cigarplug 

Great that you focus on security! As far as I know there is no Modbus standard for encrypting Modbus traffic. Therefore I'm hesitating to include a non-standard form of encryption to this project. 

But I see use in encrypting Modbus traffic. What kind of encryption did you think of? What do you think of creating a extension to the uModbus package that adds encryption and upload this package to Pypi?

jaapz, 2019-12-04:
We could include an encryption scheme for uModbus Client to an uModbus Server, maybe as a kind of middleware, or a recipe that people can use easily. 

As modbus doesn't do any encryption in the standard at all, if this is implemented we should be completely clear in documentation that it's a non-standard encryption implementation specific to uModbus.

I'm not sure if it's a great idea to add the encryption to the main package, but if we document it correctly and disable it by default we might be able to work with it.

Could be as simple as adding a layer of AES over each packet and encrypting with a default key, but how would you rotate those keys? Do we need to provide support for key rotation mechanisms, or would library users have to do that themselves? Keeping the same keys forever is only slightly better than not encrypting at all.

cigarplug, 2019-12-08:
Hi!

The traffic could be encrypted using an AEAD cypher like AES in GCM mode or chacha20-poly1305. And session keys for each connection could be generated by running an additional diffie-hellman server along with the modbus server. That is for each client connection, a new keypair must be negotiated. 

I'm not particularly bent on adding encryption to the main project. The changes could be major and  create additional dependencies such as the [cryptography](https://cryptography.io/en/latest/) python package and OpenSSL. 

The idea of publishing an extension package on PyPi sounds pretty good! But I am completely new to package development, so your assistance would be extremely good to have.