AdvancedClimateSystems/uModbus #81: ConnectionResetError: [Errno 104] Connection reset by peer

unidev10, 2019-12-10:
After `tcp.read_coils()` function, when I run the command `tcp.send_message()` I get the error - `ConnectionResetError: [Errno 104] Connection reset by peer`. 
How can this be solved?
Below is my code attached

`import socket`
`from umodbus import conf`
`from umodbus.client import tcp`
`sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)`
`sock.connect(('192.168.0.3', 102))`
`m2=tcp.read_coils(slave_id=1, starting_address=8193, quantity=1)`
`response = tcp.send_message(m2, sock)`

OrangeTux, 2019-12-11:
Hello @unidev10. You seem to want to connect to  `HOST` and there is no Modbus server running there.

My guess is that you need to replace `HOST` with a valid ip or hostname, like: 

``` python
 sock.connect(('192.168.1.1', PORT))
```

unidev10, 2019-12-11:
Hi @OrangeTux, yes I have mentioned the required ip address and the port number in my actual code. It is just for the question, I have replaced the ip address with `HOST`.

OrangeTux, 2019-12-11:
Super! I'll close the issue now. Feel free to reopen if needed.

unidev10, 2019-12-12:
Hi @OrangeTux, there has been a slight misunderstanding. In my original code, I have written `sock.connect(('192.168.0.2', 102))`. I am getting the `ConnectionResetError` for it.

jaapz, 2019-12-12:
Are you sure a modbus server is running on that IP-address? If so, there's probably something in your network filtering out your connection request, like a firewall of some kind.

The error you get is a more generic error from Python not being able to open a socket. Try to look up `ConnectionResetError` and under what circumstances Python would raise it.

I don't think this is a problem with uModbus, as uModbus doesn't do any connection handling, that's all done by the `socket` module from Python.