AdvancedClimateSystems/uModbus #82: help me

patfrench, 2020-01-02:
Hi all,

how to create an RTU slave with umodbus on the COM6 port? 
Can I query it with "minimalmodbus"?

`
#!/usr/bin/env python
from serial import Serial
from collections import defaultdict

from umodbus.server.serial import get_server
from umodbus.server.serial.rtu import RTUServer

s = Serial('COM6')
print(s)
s.timeout = 10

data_store = defaultdict(int)
app = get_server(RTUServer, s)

@app.route(slave_ids=[1], function_codes=[3, 4], addresses=list(range(0, 32)))
def read_data_store(slave_id, function_code, address):
    """" Return value of address. """
    return data_store[address]

@app.route(slave_ids=[1], function_codes=[6, 16], addresses=list(range(0, 32)))
def write_data_store(slave_id, function_code, address, value):
    """" Set value for address. """
    data_store[address] = value

write_data_store(1,16,3,784)
write_data_store(1,16,24,678)
rdata1 = read_data_store(1,4,3)
print(rdata1)
rdata2 = read_data_store(1,4,24)
print(rdata2)

if __name__ == '__main__':
    try:
        app.serve_forever()
        print(rdata1)
    finally:
        app.shutdown()
        app.server_close()`

With "minimalmodbus" I don't get any answer.
`minimalmodbus.Instrument<id=0x9abcd0, address=1, mode=rtu, close_port_after_each_call=False, precalculate_read_size=True, debug=True, serial=Serial<id=0x9ab930, open=True>(port='COM1', baudrate=9600, bytesize=8, parity='N', stopbits=1, timeout=0.05, xonxoff=False, rtscts=False, dsrdtr=False)>`

Is this the right procedure ?

Thanks

OrangeTux, 2020-01-02:
Can you post the code of your client?

patfrench, 2020-01-02:
ok !

```
import minimalmodbus

instrument = minimalmodbus.Instrument('COM1', 1)  # port name, slave address (in decimal)
value = instrument.read_register(3, 0)  # Registernumber, number of decimals
print(value)
```

> read_register(registeraddress, number_of_decimals=0, functioncode=3, signed=False)

maybe a problem connection with my converter ?

patfrench, 2020-01-03:
Hi, 
I'm sorry : problem connection between RS485 adaptator

````python
from serial import Serial
from collections import defaultdict

from umodbus.server.serial import get_server
from umodbus.server.serial.rtu import RTUServer

s = Serial('COM11')
s.timeout = 1
print(s)

data_store = defaultdict(int)
app = get_server(RTUServer, s)

@app.route(slave_ids=[1], function_codes=[3, 4], addresses=list(range(0, 32)))
def read_data_store(slave_id, function_code, address):
    """" Return value of address. """
    return data_store[address]

@app.route(slave_ids=[1], function_codes=[6, 16], addresses=list(range(0, 32)))
def write_data_store(slave_id, function_code, address, value):
    """" Set value for address. """
    data_store[address] = value
# Configuring the Data
write_data_store(1,16,1,999)
write_data_store(1,16,2,888)
write_data_store(1,16,3,777)
write_data_store(1,16,4,666)

if __name__ == '__main__':
    try:
        app.serve_forever()
        print(rdata1)
    finally:
        app.shutdown()
        app.server_close()
````

The code of my master
````
instrument.read_register(1,0)

MinimalModbus debug mode. Writing to instrument (expecting 7 bytes back): '\x01\x03\x00\x01\x00\x01ÕÊ' (01 03 00 01 00 01 D5 CA)
MinimalModbus debug mode. No sleep required before write. Time since previous read: 7419.7 ms, minimum silent period: 4.01 ms.
MinimalModbus debug mode. Response from instrument: '\x01\x03\x02\x03çøþ' (01 03 02 03 E7 F8 FE) (7 bytes), roundtrip time: 17.3 ms. Timeout setting: 1500.0 ms.

Out[121]: 999

instrument.read_register(2,0)

MinimalModbus debug mode. Writing to instrument (expecting 7 bytes back): '\x01\x03\x00\x02\x00\x01%Ê' (01 03 00 02 00 01 25 CA)
MinimalModbus debug mode. No sleep required before write. Time since previous read: 4142.0 ms, minimum silent period: 4.01 ms.
MinimalModbus debug mode. Response from instrument: '' () (0 bytes), roundtrip time: 1501.0 ms. Timeout setting: 1500.0 ms.

Traceback (most recent call last):

  File "<ipython-input-122-e029d382fb44>", line 1, in <module>
    instrument.read_register(2,0)

  File "c:\users\a7xzbzz\appdata\local\programs\python\python37-32\lib\site-packages\minimalmodbus.py", line 258, in read_register
    return self._genericCommand(functioncode, registeraddress, numberOfDecimals=numberOfDecimals, signed=signed)

  File "c:\users\a7xzbzz\appdata\local\programs\python\python37-32\lib\site-packages\minimalmodbus.py", line 697, in _genericCommand
    payloadFromSlave = self._performCommand(functioncode, payloadToSlave)

  File "c:\users\a7xzbzz\appdata\local\programs\python\python37-32\lib\site-packages\minimalmodbus.py", line 795, in _performCommand
    response = self._communicate(request, number_of_bytes_to_read)

  File "c:\users\a7xzbzz\appdata\local\programs\python\python37-32\lib\site-packages\minimalmodbus.py", line 930, in _communicate
    raise IOError('No communication with the instrument (no answer)')

OSError: No communication with the instrument (no answer)

instrument.read_register(3,0)

MinimalModbus debug mode. Writing to instrument (expecting 7 bytes back): '\x01\x03\x00\x03\x00\x01t\n' (01 03 00 03 00 01 74 0A)
MinimalModbus debug mode. No sleep required before write. Time since previous read: 5307.1 ms, minimum silent period: 4.01 ms.
MinimalModbus debug mode. Response from instrument: '\x01\x03\x02\x03\tx²' (01 03 02 03 09 78 B2) (7 bytes), roundtrip time: 17.0 ms. Timeout setting: 1500.0 ms.

Out[123]: 777

instrument.read_register(4,0)

MinimalModbus debug mode. Writing to instrument (expecting 7 bytes back): '\x01\x03\x00\x04\x00\x01ÅË' (01 03 00 04 00 01 C5 CB)
MinimalModbus debug mode. No sleep required before write. Time since previous read: 5211.8 ms, minimum silent period: 4.01 ms.
MinimalModbus debug mode. Response from instrument: '' () (0 bytes), roundtrip time: 1500.1 ms. Timeout setting: 1500.0 ms.

Traceback (most recent call last):

  File "<ipython-input-124-cf514a1cc615>", line 1, in <module>
    instrument.read_register(4,0)

  File "c:\users\a7xzbzz\appdata\local\programs\python\python37-32\lib\site-packages\minimalmodbus.py", line 258, in read_register
    return self._genericCommand(functioncode, registeraddress, numberOfDecimals=numberOfDecimals, signed=signed)

  File "c:\users\a7xzbzz\appdata\local\programs\python\python37-32\lib\site-packages\minimalmodbus.py", line 697, in _genericCommand
    payloadFromSlave = self._performCommand(functioncode, payloadToSlave)

  File "c:\users\a7xzbzz\appdata\local\programs\python\python37-32\lib\site-packages\minimalmodbus.py", line 795, in _performCommand
    response = self._communicate(request, number_of_bytes_to_read)

  File "c:\users\a7xzbzz\appdata\local\programs\python\python37-32\lib\site-packages\minimalmodbus.py", line 930, in _communicate
    raise IOError('No communication with the instrument (no answer)')

OSError: No communication with the instrument (no answer)
````

 Doesn't work every time. **problem time ?** i am on same PC

Thanks