AdvancedClimateSystems/uModbus #84: UnicodeDecodeError from setup.py with python3.6 and LANG=C

ktns, 2020-02-05:
Hi, 

I've encountered an error while installing your package in a docker image.
It boils down like below.
```
%docker run --rm python:3.6 env LANG=C pip install --no-binary :all: uModBus==1.0.3
Collecting uModBus==1.0.3
  Downloading uModbus-1.0.3.tar.gz (19 kB)
    ERROR: Command errored out with exit status 1:
     command: /usr/local/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-7mj5qq3n/uModBus/setup.py'"'"'; __file__='"'"'/tmp/pip-install-7mj5qq3n/uModBus/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-7mj5qq3n/uModBus/pip-egg-info
         cwd: /tmp/pip-install-7mj5qq3n/uModBus/
    Complete output (7 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-7mj5qq3n/uModBus/setup.py", line 12, in <module>
        long_description = open(os.path.join(cwd, 'README.rst'), 'r').read()
      File "/usr/local/lib/python3.6/encodings/ascii.py", line 26, in decode
        return codecs.ascii_decode(input, self.errors)[0]
    UnicodeDecodeError: 'ascii' codec can't decode byte 0xce in position 547: ordinal not in range(128)
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

I think I can prepare a PR to fix this.

OrangeTux, 2020-02-05:
Thanks for opening this issue. It seem a duplicate of #60 . I was wondering, could you share the Dockerfile you used? I'd like to reproduce the problem myself.

jaapz, 2020-02-08:
@OrangeTux i can reproduce this by running the command in the original issue.

```bash
$ docker run --rm python:3.6 env LANG=C pip install --no-binary :all: uModBus==1.0.3
Collecting uModBus==1.0.3
  Downloading https://files.pythonhosted.org/packages/eb/1c/852af30263b6fd44e160951d35f70f26a4c7b1f15458c174402c04d0753b/uModbus-1.0.3.tar.gz
    ERROR: Complete output from command python setup.py egg_info:
    ERROR: Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-qagdmmru/uModBus/setup.py", line 12, in <module>
        long_description = open(os.path.join(cwd, 'README.rst'), 'r').read()
      File "/usr/local/lib/python3.6/encodings/ascii.py", line 26, in decode
        return codecs.ascii_decode(input, self.errors)[0]
    UnicodeDecodeError: 'ascii' codec can't decode byte 0xce in position 547: ordinal not in range(128)
    ----------------------------------------
ERROR: Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-qagdmmru/uModBus/
WARNING: You are using pip version 19.1.1, however version 20.0.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

When you remove the `env LANG=C` part from the command the build works.

A quick google gives me this: https://stackoverflow.com/questions/39206705/python-3-throws-unicodeencodeerror-with-environment-variable-lang-c

tldr: setting LANG=C tells python to use ASCII as output encoding. However the README.rst contains unicode characters, so it breaks.

I think setting `encoding="utf-8"` in the `open` call should fix this:

```python
long_description = open(os.path.join(cwd, 'README.rst'), 'r', encoding="utf-8").read()
```

I haven't tested this though.

jaapz, 2020-02-08:
Ah I see that's also what you proposed in the PR :upside_down_face: