AdvancedClimateSystems/uModbus #90: Invalid error codes

ovunccetin, 2020-04-19:
According to [Modbus Application Protocol V1.1b3](http://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf) error codes for "GATEWAY PATH UNAVAILABLE" and "GATEWAY TARGET DEVICE FAILED TO RESPOND" are 10 and 11, respectively. However, they are 11 and 12 in exceptions module.

jaapz, 2020-04-23:
Oops! Would you be able to open a pull request?

jacobm32, 2020-06-02:
any update on this?
how to send error code 11 or 12 (from tcp server), 
i am trying to make modbus gateways (umodbus_rtu_master read data and send it to umodbus_tcp_server )
so if umodbus_rtu_master dont get any data(due to connection problem and timeout), how to send error code "GATEWAY TARGET DEVICE FAILED TO RESPOND" from umodbus_tcp_server

kindly guide..(with example code, if possible, )

PR #94 (2020-06-23): Fix 2 error codes
The error codes where off by 1.

Fixes: #90
commit 5fdc3b8f (2020-06-23): Fix 2 error codes

The error codes where off by 1.

Fixes: #90
closed 2020-06-23 by jaapz