AdvancedClimateSystems/uModbus #91: Modbus RTU over TCP

azat385, 2020-05-21:
Hello!

**Is it possible to use rtu frame over TCP connection?**
Slave device is connected to pc via moxa nport 5150 (which is changing just interface but not the messages)

Thanks a lot!

tiagocoutinho, 2020-11-07:
@azat385 ,

You can create a Serial object which talks to moxa via raw socket or even [RFC 2217](https://tools.ietf.org/html/rfc2217) (I think moxa supports both in all models).
I recommend you use raw socket unless you need to change the serial line settings frequently (like using different baudrate)

Here is an example:

```python
import serial

serial_port = serial.serial_for_url("socket://moxa.acme.org:6610")
message = rtu.write_multiple_coils(slave_id=1, address=1, values=[1, 0, 1, 1])
response = rtu.send_message(message, serial_port)

serial_port.close()
```

Hope it helps.

azat385, 2020-12-07:
@tiagocoutinho thanks a lot !

closed 2020-12-07 by azat385