AdvancedClimateSystems/uModbus #96: Can this be used for Modbus sniffing?

vshymanskyy, 2020-07-10:
One interesting application of this could be to be able to attach to an existing Modbus connection (just the RX line), and to be able to decode/display the messages.
Wondering if any additional development is needed to achieve this.

OrangeTux, 2020-07-13:
I guess you want to sniff on Modbus RTU communication, correct? That would be possible. 

I think I would override `RTUServer.execute_route()` to decode and display the messages.

``` python
def execute_route(self, meta_data, request_pdu):
    function = create_function_from_request_pdu(request_pdu)
    print(function)
```

OrangeTux, 2020-07-20:
Closed due to inactivity. Feel free to reopen if needed.

closed 2020-07-20 by OrangeTux