AdvancedClimateSystems/uModbus #97: Redundancy Check issues

dmhouston95, 2020-07-22:
Function **validate_crc** does not evaluate the CRC bytes correctly. Incorrect use of **not** and **==**

OrangeTux, 2020-07-23:
What do you mean? Instead of 

``` python
    if not struct.unpack('<H', get_crc(msg[:-2])) ==\
            struct.unpack('<H', msg[-2:]):
        raise CRCError('CRC validation failed.')
```

you suggest:

``` python
    if  struct.unpack('<H', get_crc(msg[:-2])) !=\
            struct.unpack('<H', msg[-2:]):
        raise CRCError('CRC validation failed.')

dmhouston95, 2020-07-23:
Yes. The expression evaluates false because **not** of an integer is always false unless the integer is zero. And then if you are trying to compare a boolean false (evaluated from the **not** expression) with an integer that will also always be false.

dmhouston95, 2020-07-23:
This would also be vaild
`if not (struct.unpack('<H', get_crc(msg[:-2])) ==struct.unpack('<H', msg[-2:])):
        raise CRCError('CRC validation failed.')`

rgov, 2020-07-24:
See operator precedence: https://docs.python.org/3/reference/expressions.html#operator-precedence

`not` is less binding than `==` so I believe that `not a == b` is (in most cases?) the same as `a != b`. Since `==` and `is` have the same precedence, this is easier to see this way:

```
>>> not 1
False
>>> not 1 is True
True
```

If `not` bound more tightly, then it would be the same as `(not 1) is True` which doesn't match the behavior. But instead we get `not (1 is True)`.

However `!=` reads better to me.