AdvancedClimateSystems/uModbus #107: Read Multiple Hosts

shnabin, 2020-10-06:
Hi, 

I need help with the following 2 points. 

1. Is it possible to connect to multiple hosts in one instance,  (Ex: Server IP : 192.168.1.1 , Client IP : 192.168.1.2, 192.168.1.3, etc), If its possible kindly provide an example. 
2. From point 1, once I read the data from multiple clients, can I act as modbus server/gateway and give the data to other devices.  
Ex: 192.168.1.2 (host/client) --> (server) 192.168.1.1 (client/host) --> 192.168.10.2. If its possible kindly provide an example. 

Appreciate your help. Thank you !!

OrangeTux, 2020-10-15:
1) Yes. Multiple clients can connect to the same server. Just copy the `[sample client](https://github.com/AdvancedClimateSystems/uModbus/blob/master/scripts/examples/simple_tcp_client.py) to all your hosts and run them.
2) Yes, that is also possible.

jaapz, 2020-10-19:
2. is possible but you'll have to write some "glue code" yourself. you can set up a umodbus server that handles reads and writes and proxies those to another umodbus client instance

Currently we don't have example code for this but if you understand how the server and client work, that should be relatively trivial to implement.