AdvancedClimateSystems/uModbus #108: ServerDeviceFailureError: An unrecoverable error occurred.

princeBruce, 2020-10-08:
Hi there, when I try to use the package as a client to connect to a slave device, it throws an error as in the title. Have used other Modbus TCP master utility to successfully connect to this device, and in the meanwhile, this package can be used to connect to other slaves without issues. Can I get some help or clues regarding this exception? Thanks much!

Bruce

OrangeTux, 2020-10-09:
Can you provide the request you're sending?

princeBruce, 2020-10-09:
Please see below. The socket can be established without any issue, but the exception arise when sending the message. Similar exception gets thrown when sending a read request.
----
import socket
from umodbus import conf
from umodbus.client import tcp
sock2 = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
sock2.connect(('192.111.1.12', 502))    
message = tcp.write_single_register(slave_id=1, address=12, value=200)
response = tcp.send_message(message, sock2)

princeBruce, 2020-10-12:
More information is added here.

**Running code:**

    message = tcp.read_holding_registers(slave_id=1, starting_address=12, quantity=1)
    print(message)
    response = tcp.send_message(message, sock2)

**Message sent:**

b'\xad\xb8\x00\x00\x00\x06\x01\x03\x00\x0c\x00\x01'

**Error Message:**
Traceback (most recent call last):

  File "C:\Users\lg138523sd\.spyder-py3\temp.py", line 57, in <module>
    response = tcp.send_message(message, sock1)

  File "C:\Users\lg138523sd\Anaconda3\lib\site-packages\umodbus\client\tcp.py", line 262, in send_message
    raise_for_exception_adu(response_error_adu)

  File "C:\Users\lg138523sd\Anaconda3\lib\site-packages\umodbus\client\tcp.py", line 247, in raise_for_exception_adu
    pdu_to_function_code_or_raise_error(resp_pdu)

  File "C:\Users\lg138523sd\Anaconda3\lib\site-packages\umodbus\functions.py", line 118, in pdu_to_function_code_or_raise_error
    raise error_code_to_exception_map[error_code]

ServerDeviceFailureError: An unrecoverable error occurred.

OrangeTux, 2020-10-13:
Without other information I would say: The problem is with your device. But considering your statement "Have used other Modbus TCP master utility to successfully connect to this device" I'm less sure. 

Is there any chance that you can use a different TCP master to execute the same request as you try to execute with uModbus and than capture the traffic of this different TCP master? Than I can compare messages that uModbus produces with that message from the different TCP master.

princeBruce, 2020-10-13:
Hi, have figured it out. It's because of the reference address offset. The address specified in the device is offset by one, so that I actually need to specify the address as 11 (0b) in uModbus to visit the register at address "12". The other master utility happens to use the same address offset convention so it works.

Besides, the value I try to read/write is a 32 bit float so it takes 2 16-bit register starting from 0b. I guess because of this, if I use 0c in the uModbus request packet, it triggers an error instead of returning a value, though that will not be what I want. After using the correct address and APIs for multiple registers, it works all right.  

Thanks again for you attention and help!

OrangeTux, 2020-10-13:
Great that you found it out. That Modbus offset of 1 is very confusing. I'm honestly wonder why that was added to the Modbus specification.