AdvancedClimateSystems/uModbus #111: How do you access the raw (bytes) response for decoding messages that aren't composed of 16-bit integers?

fake-name, 2021-01-20:
Effectively the title. I have a device that packs 32-bit floats into two sequential registers. 

Currently, there seems to be no way to get anything *but* the already decoded 16-bit integer sequence. 

I can re-pack the return data from `umodbus.client.serial.rtu.send_message()` into a byte-array, and then re-decode that using the device's structure, bit that seems asinine given that all I really need is uModbus to not do the (in this application incorrect) decode step.

I spent a bit of time looking into potentially just replacing `send_message()` with a version that still does the CRC check and message trimming, but not the decode, but it appears that separating the message validation and trimming from the decoding is sufficiently annoying that I can't see this as anything other then a bug.

How do I get the raw response the modbus device returns without having to re-pack the data into a bytearray?

azat385, 2021-06-30:
is it possible to get response ADU from `send_message`?

https://github.com/AdvancedClimateSystems/uModbus/blob/f1128a73e43f565bacedd1ae99d077d7c9c831f3/umodbus/client/serial/rtu.py#L205

johanvdw, 2022-06-22:
I also think it would be a good idea to allow setting the datatype in send_message, to allow decoding into something different than uint16 or int16 (depending on conf.SIGNED_VALUES).

In general I actually think that setting conf.SIGNED_VALUES is not a good practice, because changing that value globally is not ideal when you want to support different devices.