AdvancedClimateSystems/uModbus #123: Possible typo in functions.py

ardeley, 2022-10-12:
I have not destruction-tested, but I am worried by what seems to be a typo in line 1510 (in class WriteMulipleRegisters). The maximum limit is set to 0x7B0 - but I believe what is intended is 0x007B. The MODBUS Application Protocol page 30 seems clear.
I think code may have been copied and pasted from line 1300 (in class WriteMultipleCoils).

ardeley, 2022-10-19:
As a week has gone by with no comment, I now attach a copy of pages 30-31 of the MODBUS Application Protocol Specification V1.1b3. I think there is enough information on these pages for anyone to conclude that line 1510 (in class WriteMultipleRegisters) of functions.py applies the wrong test: the maximum limit of the test should be 0x007B.
This seems to need prompt attention.
[pages30+31_MODBUSprotocol.pdf](https://github.com/AdvancedClimateSystems/uModbus/files/9818699/pages30%2B31_MODBUSprotocol.pdf)

acolomb, 2022-10-19:
I seriously doubt there will be any reaction from the maintainers. I sent a couple of issues and pull requests one or two years ago without getting any feedback whatsoever since then. Have a look at the issues list, I guess this project is not actively maintained anymore. My solution was to just patch in what I needed and deploy a modified version as Python wheel.

ardeley, 2022-10-19:
@acolomb, I think you are probably right. I have built my own .php code to use MODBUS safely in a web-app (with myself as the principal destructive client!). Strange that GitHub, Advanced Climate Systems and @OrangeTux continue to associate their names with a Python module, when it includes a false test which could mess up MODBUS. I am grateful, however, for insights into coding for this protocol.

acolomb, 2022-10-19:
Well, the issue is recorded here. Whoever might decide to fork this project and pick up maintainership, they can go through the open issues list and easily apply your simple fix.